Xilinx/DPU-PYNQ #1: Problems when importing runner VART module in python3.6

jlamperez, 2020-05-09:
Hi @yunqu ,

First of all thank you for your effort supporting DPU on Pynq boards it is an incredible work!

I've been able to run different dpu notebooks but now when I try to load models using VART and `libvart-dpu-runner.so` like in this example, [resnet50_mt_py](https://github.com/Xilinx/Vitis-AI/tree/master/VART/samples/resnet50_mt_py), I am having problems when running `import runner`. It says that:

> ModuleNotFoundError: No module named 'runner'

I've been digging and I have seen that *runner.so* is installed in */usr/lib/python3.5/site-packages/runner.so* but notebooks use Python3.6. Could this *runner.so* be installed/supported in Python 3.6?

Best regards,
Jorge.

yunqu, 2020-05-11:
@jlamperez  Currently we don't have good support for VAI run-time. The main reason is what you found - VAI is depending on python3.5 but we have python3.6. So we only support DNNDK run-time to use DPU.

jlamperez, 2020-05-11:
@yunqu Okey, no problem. Thank you very much for your answer!

closed 2020-05-11 by jlamperez