Xilinx/DPU-PYNQ #5: Model deployment

zackdilan, 2020-06-08:
Hello, 
Is there any examples showing how to deploy the compiled models on the Ultra96 board. If not where can I find it? I have seen the zcu104/2 board examples using Vitis AI library and VART. If I want to do the same with Ultra96, is it possible?

yunqu, 2020-06-08:
Check our training/model compilation instructions: https://github.com/Xilinx/DPU-PYNQ/tree/master/host

zackdilan, 2020-06-09:
Thanks, I have found the notebooks and they are pretty good. My actual question was, Is it possible to run a similar application that is available in https://github.com/Xilinx/Vitis-AI/tree/v1.1/mpsoc/vitis_ai_dnndk_samples ? I tried to compile the SSD detection model with VItis AI for ultra96 as you have mentioned and tried to deploy the model in the board with the source code https://github.com/Xilinx/Vitis-AI/tree/v1.1/mpsoc/vitis_ai_dnndk_samples/video_analysis as here. I have did the same application with earlier tools (DNNDK ) but I am not able to run the application now with the pynq dpu on board I was able to compile the application sucessfully without any errors.
it shows `[DNNDK_XRT] Cannot find device, index: 0
`
what could be the reason?

yunqu, 2020-12-20:
Make sure you upgraded all the things required in the `upgrade` folder

closed 2020-12-20 by yunqu