Xilinx/DPU-PYNQ #6: Pynq-DPU install takes a lot of time

zackdilan, 2020-06-08:
I am facing a lot of time delay in installing pynq dpu . Is it normal or not? I was able to successfully build the pynq upgrade. Please let me know.

yunqu, 2020-06-08:
Which step are you referring to exactly? How much time does it take?

zackdilan, 2020-06-08:
the pip install pynq-dpu step. 
[image]

yunqu, 2020-06-08:
Is this based on pynq 2.5 image and you have run our upgrade? It looks like you do not have proper pynq package installed.

zackdilan, 2020-06-08:
Yes this is based on pynq 2.5 image and done the pynq image upgrade successfully.

yunqu, 2020-06-08:
We have tested the whole flow based on a clean 2.5 image. I did not see this issue; so maybe you changed something on the image?

zackdilan, 2020-06-09:
Yes, there were some problems with my image, I tried to do the whole process again and it worked. Thanks

closed 2020-06-09 by zackdilan