Xilinx/DPU-PYNQ #10: dexplorer output

jlamperez, 2020-07-16:
Hi,

When working with `dpu_inception_v1.ipynb` I have seen that `!dexplorer -w` has the next output:

```
[DPU IP Spec]
IP  Timestamp            : 2020-03-26 13:30:00
DPU Core Count           : 1

[DPU Core Configuration List]
DPU Core                 : #0
DPU Enabled              : Yes
DPU Arch                 : B1600
DPU Target Version       : v1.4.1
DPU Freqency             : 300 MHz
Ram Usage                : Low
DepthwiseConv            : Enabled
DepthwiseConv+Relu6      : Enabled
Conv+Leakyrelu           : Enabled
Conv+Relu6               : Enabled
Channel Augmentation     : Disabled
Average Pool             : Enabled
```

I am a little bit confuse about this output because I have created another dpu design with other DPU frequencies and another architectures and when I tried `dexplorer` after loading the overlay, all the time the output is the same, it doesn't change.

Is this `dexplorer` output hardcoded?

Also I was having reboot problems when using designs with 300/600 MHz frequency designs, that I am not having when I use the default design so how it is possible to be DPU Freqency equal to 300MHz?

yunqu, 2020-07-16:
Perhaps a better place to post this kind of question is discuss.pynq.io. But to answer your question, that 300MHz is the DPU frequency, so its 150/300MHz. On Ultra96, we have a power issue when using DPU, which is a known issue. We had to reduce the clock frequency. As a matter of fact, the best you could do, based on my experiments, is around 190/380MHz. Any frequency above that will cause a reboot.

For the 1st question, no, it is not hard-coded. But to debug your problem we need more detail. So perhaps you want to post that on discuss.pynq.io.

jlamperez, 2020-07-16:
Okey I will change to discuss.pynq.io.

Thanks!

closed 2020-07-16 by jlamperez

jlamperez, 2020-07-16:
If someone is interested [here](https://discuss.pynq.io/t/dpu-pynq-dexplorer-output/1507) discuss.pynq.io thread