Xilinx/DPU-PYNQ #15: Self rebuild ZCU104 DPU kernel died

yluo39github, 2020-10-28:
I follow the instructions with the pre-build HW (dpu.bit dpu.hwh, dpu.xclbin) on ZCU104; it implements successfully.

And I follow the Rebuild instructions to generate my own HW (dpu.bit dpu.hwh, dpu.xclbin), while implement on the example, it shows the  "Kernel Restarting The kernel appears to have died. It will restart automatically."

Can anyone tell me what the problem is?

yunqu, 2020-10-28:
You can check /var/log/jupyter.log for the real reason why the kernel is died.

yluo39github, 2020-10-29:
Thank you for your reply, here attached my /var/log/jupyter.log

[I 15:38:21.301 NotebookApp] [jupyter_nbextensions_configurator] enabled 0.4.1
[I 15:38:21.827 NotebookApp] JupyterLab extension loaded from /usr/local/lib/pytthon3.6/dist-packages/jupyterlab
[I 15:38:21.828 NotebookApp] JupyterLab application directory is /usr/local/sharre/jupyter/lab
[I 15:38:21.861 NotebookApp] Serving notebooks from local directory: /home/xilinnx/jupyter_notebooks
[I 15:38:21.861 NotebookApp] 0 active kernels
[I 15:38:21.861 NotebookApp] The Jupyter Notebook is running at:
[I 15:38:21.861 NotebookApp] http://0.0.0.0:9090/
[I 15:38:21.862 NotebookApp] Use Control-C to stop this server and shut down all kernels (twice to skip confirmation).
[I 15:38:21.872 NotebookApp] 302 GET / (127.0.0.1) 2.59ms
[I 15:38:21.878 NotebookApp] 302 GET /tree? (127.0.0.1) 4.10ms
[I 15:38:51.602 NotebookApp] 302 GET / (192.168.2.2) 2.04ms
[W 15:38:52.513 NotebookApp] 404 GET /i18n/zh-CN/LC_MESSAGES/nbjs.json?v=202010229153819 (192.168.2.2) 65.27ms referer=http://192.168.2.99:9090/tree?
[W 15:38:52.801 NotebookApp] 404 GET /static/components/moment/locale/zh-cn.js?vv=20201029153819 (192.168.2.2) 14.26ms referer=http://192.168.2.99:9090/tree?
[W 15:38:57.637 NotebookApp] 404 GET /static/components/moment/locale/zh-cn.js?vv=20201029153819 (192.168.2.2) 6.27ms referer=http://192.168.2.99:9090/tree/pynq--dpu
[W 15:38:57.651 NotebookApp] 404 GET /i18n/zh-CN/LC_MESSAGES/nbjs.json?v=202010229153819 (192.168.2.2) 6.13ms referer=http://192.168.2.99:9090/tree/pynq-dpu
[W 15:39:05.643 NotebookApp] 404 GET /i18n/zh-CN/LC_MESSAGES/nbjs.json?v=202010229153819 (192.168.2.2) 8.02ms referer=http://192.168.2.99:9090/notebooks/pynq-dpuu
/dpu_yolo_v3.ipynb
[W 15:39:05.653 NotebookApp] 404 GET /static/components/moment/locale/zh-cn.js?vv=20201029153819 (192.168.2.2) 8.04ms referer=http://192.168.2.99:9090/notebooks//pynq-dpu/dpu_yolo_v3.ipynb
[I 15:39:06.182 NotebookApp] Kernel started: 6294dded-be3e-49d3-862c-42bc0c142700e
[I 15:39:09.288 NotebookApp] Adapting to protocol v5.1 for kernel 6294dded-be3e--49d3-862c-42bc0c14270e
[I 15:41:06.156 NotebookApp] Saving file at /pynq-dpu/dpu_yolo_v3.ipynb
[DNNDK] DPU timeout while execute DPU Task:tf_yolov3-0
[DPU mode]
normal

[DPU timeout limitation (in seconds)]
10

[DPU Debug Info]
Core 0 schedule : 1
Core 0 interrupt: 1
Core 1 schedule : 0
Core 1 interrupt: 0

[DPU Resource]
DPU Core        : 0
State           : Idle
PID             : 4085
TaskID          : 4085
Start           : 13824
End             : 5032828

DPU Core        : 1
State           : Idle
PID             : 0
TaskID          : 0
Start           : 0
End             : 0

[DPU Core 0 Register]
CTL       : 0x00000001
GIE       : 0x00000000
IRQ       : 0x00000000
HP        : 0x07070f0f
CODE      : 0x0000000000060300
BASE0     : 0x0000000060400000
BASE1     : 0x0000000063f00000
BASE2     : 0x0000000000000000
BASE3     : 0x0000000000000000
BASE4     : 0x0000000000000000
BASE5     : 0x0000000000000000
BASE6     : 0x0000000000000000
BASE7     : 0x0000000000000000
CYCLE_H   : 0x00000000
CYCLE_L   : 0x00000000
REGVER    : 0x32016b02
TIMESTAMP : 0x1431a0d2
GITID     : 0x07b7683a
GITTIME   : 0x77675797
VERSION   : 0x00000141
TIMER     : 0x00000000
ARCH      : 0x31381010
RAM       : 0x00001333
LOAD      : 0x00000102
CONV      : 0x00000111
SAVE      : 0x00000002
POOL      : 0x00000001
ELEW      : 0x00000001
DWCV      : 0x00000014
MISC      : 0x00000001
DPU STATUS: 0x00000000
AXI STATUS: 0x00015454
LOAD START: 26
LOAD END  : 26
SAVE START: 0
SAVE END  : 0
CONV START: 2
CONV END  : 1
MISC START: 0
MISC END  : 0

[DPU Core 1 Register]
CTL       : 0x00000004
GIE       : 0x00000000
IRQ       : 0x00000000
HP        : 0x00000000
CODE      : 0x0000000000000000
BASE0     : 0x0000000000000000
BASE1     : 0x0000000000000000
BASE2     : 0x0000000000000000
BASE3     : 0x0000000000000000
BASE4     : 0x0000000000000000
BASE5     : 0x0000000000000000
BASE6     : 0x0000000000000000
BASE7     : 0x0000000000000000
CYCLE_H   : 0x00000000
CYCLE_L   : 0x00000000
REGVER    : 0x32016b02
TIMESTAMP : 0x1431a0d2
GITID     : 0x07b7683a
GITTIME   : 0x77675797
VERSION   : 0x00000141
TIMER     : 0x00000000
ARCH      : 0x31381010
RAM       : 0x00001333
LOAD      : 0x00000102
CONV      : 0x00000111
SAVE      : 0x00000002
POOL      : 0x00000001
ELEW      : 0x00000001
DWCV      : 0x00000014
MISC      : 0x00000001
DPU STATUS: 0x00000000
AXI STATUS: 0x00015454
LOAD START: 0
LOAD END  : 0
SAVE START: 0
SAVE END  : 0
CONV START: 0
CONV END  : 0
MISC START: 0
MISC END  : 0

[I 15:41:15.171 NotebookApp] KernelRestarter: restarting kernel (1/5), keep randdom ports 
kernel 6294dded-be3e-49d3-862c-42bc0c14270e restarted

yluo39github, 2020-11-18:
I compare ZCU104_base and ZCU104_dpu platform. Some pins are different, so which platform are you use?

yluo39github, 2020-11-20:
I found the program died while I call: n2cube.dpuLoadKernel

yunqu, 2020-12-20:
If you change the DPU configuration you will have to recompile your model.

closed 2020-12-20 by yunqu