Xilinx/DPU-PYNQ #22: Compatibility with Pynq-Z2

Sellto, 2021-01-15:
Hello
I could see that vitis ai 1.3 was available and I saw in an issue that it was a constraint to have pynq-DPU which would be compatible with the Z2. Can you estimate the release date of a version for the pynq Z2?

Sellto, 2021-02-02:
No news?

yunqu, 2021-02-04:
@Sellto  We are still working on the aarch64 but we hope we will kick off armv7l support very soon.

yunqu, 2021-03-02:
There are some issues with support on zynq 7000. Vitis AI 1.3.1 will support Zynq 7000 so we will wait on that release to update our repo.

yunqu, 2021-03-24:
We have internally Zynq7000 support. However, at this point due to lack of man power to support it, we will not release it publicly. The basic steps are similar to Zynq Ultrascale, since all the source code is available on Vitis AI git. If you still need it I can send over a private compiled VART library tarball; and as a matter of fact, that could be compiled by users by themselves as well.

closed 2021-03-24 by yunqu

chebro, 2022-02-10:
> If you still need it I can send over a private compiled VART library tarball; and as a matter of fact, that could be compiled by users by themselves as well.

could you please point to the instructions to compile and run pynq-dpu on the pynq z2?