Xilinx/DPU-PYNQ #24: Error in train_mnist_model notebook

balintmaci, 2021-01-21:
After setting up the Vitis AI CPU docker container, I ran it from host/ using `./docker_run.sh xilinx/vitis-ai-cpu:latest` following the guide in host/README.md

In the container I ran the compile.sh once, so it copies the board files, then the other commands to start the notebook
```
conda activate vitis-ai-tensorflow
./compile.sh Ultra96 tf_yolov3_voc_416_416_65.63G_1.3
yes | pip install matplotlib keras==2.2.5
jupyter notebook --ip=0.0.0.0 --port=8080
```
I had to slightly modify the compile.sh to run where the MODEL_NAME is referenced with normal braces instead of curly braces.

I also tried to manually copy the board files and had the same result.

The notebook mostly runs fine, there are some warnings about deprecated functions, but everything works until the very last part. I modified it slightly to point to the Ultra96 board images instead of the default.
```
!vai_c_tensorflow \
    --frozen_pb quantized/deploy_model.pb \
    --arch /opt/vitis_ai/compiler/arch/DPUCZDX8G/Ultra96/Ultra96.json \
    --output_dir . \
    --net_name mnist_classifier
```
result:
```
**************************************************
* VITIS_AI Compilation - Xilinx Inc.
**************************************************
[INFO] Namespace(inputs_shape=None, layout='NHWC', model_files=['quantized/deploy_model.pb'], model_type='tensorflow', out_filename='./mnist_classifier_org.xmodel', proto=None)
[INFO] tensorflow model: quantized/deploy_model.pb
[INFO] parse raw model     : 11%| | 1/9 [00:00<00:00, 5857.97it/s]              
Traceback (most recent call last):
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/bin/xnnc-run", line 33, in <module>
    sys.exit(load_entry_point('xnnc==1.3.0', 'console_scripts', 'xnnc-run')())
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/xnnc/__main__.py", line 194, in main
    normal_run(args)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/xnnc/__main__.py", line 178, in normal_run
    in_shapes=in_shapes if len(in_shapes) > 0 else None,
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/xnnc/xconverter.py", line 131, in run
    xmodel = CORE.make_xmodel(model_files, model_type, _layout, in_shapes)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/xnnc/core.py", line 104, in make_xmodel
    model_files, layout, in_shapes=in_shapes, model_type=model_t
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/xnnc/translator/tensorflow_translator.py", line 97, in to_xmodel
    model_name, raw_nodes, layout, in_shapes, model_fmt, model_type
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/xnnc/translator/tensorflow_translator.py", line 161, in create_xmodel
    xmodel = cls.__create_xmodel_from_tf1(name, layers, layout, in_shapes)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/xnnc/translator/tensorflow_translator.py", line 243, in __create_xmodel_from_tf1
    xmodel_name, layout, layers, const_layer_dict, super_const_dict, in_shapes
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/xnnc/translator/tensorflow_translator.py", line 1847, in __generate_xmodel
    ), f"[ERROR] TF Conv2d requires two inputs: actual: {bottom}."
AssertionError: [ERROR] TF Conv2d requires two inputs: actual: ['input_data'].
```

Any ideas what's causing the error? Anything I can do to mitigate it?

balintmaci, 2021-01-25:
issue also doesn't seem to be present with the older version docker dontainer

gjkunde, 2021-01-26:
My understanding is that "latest" runs Vitis 1.3 ... all the workflows including PYNQ-DPU are for 1.2,
so you should start with:.

/docker_run.sh xilinx/vitis-ai:1.2.82

in that case it produces (when fixing the legacy DPUV2 in the json file call):
**************************************************
* VITIS_AI Compilation - Xilinx Inc.
**************************************************
[VAI_C][Warning] layer [output_logits_Softmax] (type: Softmax) is not supported in DPU, deploy it in CPU instead.

Kernel topology "mnist_classifier_kernel_graph.jpg" for network "mnist_classifier"
kernel list info for network "mnist_classifier"
                               Kernel ID : Name
                                       0 : mnist_classifier_0
                                       1 : mnist_classifier_1

                             Kernel Name : mnist_classifier_0
--------------------------------------------------------------------------------
                             Kernel Type : DPUKernel
                               Code Size : 4.48KB
                              Param Size : 0.66MB
                           Workload MACs : 1.78MOPS
                         IO Memory Space : 6.05KB
                              Mean Value : 0, 0, 0, 
                      Total Tensor Count : 4
                Boundary Input Tensor(s)   (H*W*C)
                         input_data:0(0) : 28*28*1

               Boundary Output Tensor(s)   (H*W*C)
               output_logits_MatMul:0(0) : 1*1*10

                        Total Node Count : 3
                           Input Node(s)   (H*W*C)
                 conv2d_1_convolution(0) : 28*28*1

                          Output Node(s)   (H*W*C)
                 output_logits_MatMul(0) : 1*1*10

                             Kernel Name : mnist_classifier_1
--------------------------------------------------------------------------------
                             Kernel Type : CPUKernel
                Boundary Input Tensor(s)   (H*W*C)
              output_logits_Softmax:0(0) : 1*1*10

               Boundary Output Tensor(s)   (H*W*C)
              output_logits_Softmax:0(0) : 1*1*10

                           Input Node(s)   (H*W*C)
                   output_logits_Softmax : 1*1*10

                          Output Node(s)   (H*W*C)
                   output_logits_Softmax : 1*1*10

balintmaci, 2021-01-26:
I arrived to the same conclusion, can confirm

closed 2021-01-26 by balintmaci

Afef00, 2022-01-28:
Am facing the same issue.
The problem is elf models are not supported anymore and only xmodels are supported!! So is there any updated solution for this problem ??