Xilinx/DPU-PYNQ #26: Ultra96 dcf file ?

gjkunde, 2021-01-25:
Hello,

the ULTRA96 json file references:

    "target"   : "DPUCZDX8G",
    "dcf"      : "/opt/vitis_ai/compiler/arch/DPUCZDX8G/Ultra96/Ultra96.dcf",
    "cpu_arch" : "arm64"

alas Vitis 1.2 does not have an ULTRA96 directory in /opt/vitis_ai/compiler/arch/DPUCZDX8G/
it appears necessary to not only post the pynqdpu.dpu.ultra96.hwh file but the dcf as well ?

Thank you for your support, Gerd

yunqu, 2021-01-25:
It should be generated when running the script: https://github.com/Xilinx/DPU-PYNQ/blob/master/host/compile.sh#L37

gjkunde, 2021-01-25:
Thank you for the quick response, I am on the quest to run 

https://github.com/Xilinx/DPU-PYNQ/blob/master/host/train_mnist_model.ipynb, so I did not use the 
compile.sh script for that seems to be designed for the model zoo zip file.

I see the "dlet -f" command there, just did not expect them in the compile script :-)

balintmaci, 2021-01-26:
I recommend that you run compile.sh before running the notebook just so it copies the required files, that's the quickest way

closed 2021-01-26 by gjkunde