Xilinx/DPU-PYNQ #30: Debug mode for dpu_mnist_classifier.ipynb

gjkunde, 2021-01-26:
I read in UG1414 about debugging ncube:

_It involves the following five steps.
1. Run Vitis AI quantizer to generate the golden baseline from the quantized model.
2. Build the model as debug mode DPU kernel by Vitis AI compiler with option --dump
fused_graph_info specified.
3. Before launching the running of DPU application, run command dexplorer -m debug to
switch runtime N2Cube into debug mode, or calling dpuEnableTaskDebug() to enable
debug mode for the dedicated DPU task only (other tasks will not be affected).
4. Run the DPU application and get the raw dump data for DPU task’s each node.
5. Compare DPU raw dump data with the golden baseline from quantizer_

As it turns out the kernel dies even if there is just a typo in the nodes for ncube.
It would be really good to get access to that feature.

I tried to at the line:
**!dexplorer -m**
in the nodebook, but that seems to spawn its own shell and does not affect the notebook.

I tried 
    n2cube.dpuSetInputTensorInHWCFP32(task, KERNEL_CONV_INPUT, test_data[j], input_len)
    **n2cube.dpuEnableTaskDebug(task)**
    n2cube.dpuRunTask(task)

again no effect, would you please so kind to point out how to get maximum debugging into from ncube ?

Thank you so much for your help, Gerd

schelleg, 2021-05-05:
We might not be the best to resolve VAI debugging ... perhaps ask over at the Vitis-AI GitHub?

schelleg, 2021-05-14:
closing due to inactivity ... please reopen as needed.

closed 2021-05-14 by schelleg