Xilinx/DPU-PYNQ #31: Something changed today (keras 2.2.5 ?)  - cannot run training_mnist_model.ipynb anymore

gjkunde, 2021-01-29:
Hello,

I was working with this notebook as a template to understand the workflow,
alas as of today it does not run anymore. I am running:

./docker_run.sh xilinx/vitis-ai:1.2.82

and within

pip install keras 2.2.5

First indication is that history.history does not have 'acc' anymore but 'accuracy' instead,
so the plots fail. 

That seems to be a feature of  keras 2.3.x ?
_According to the 2.3.0 Release Notes: 
"Metrics and losses are now reported under the exact name specified by the user (e.g. if you pass metrics=['acc'], your metric will be reported under the string "acc", not "accuracy", and inversely metrics=['accuracy'] will be reported under the string "accuracy"."_

**But most importantly I get for the freeze:**
Traceback (most recent call last):
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/bin/freeze_graph", line 10, in <module>
    sys.exit(run_main())
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/tensorflow_core/python/tools/freeze_graph.py", line 487, in run_main
    app.run(main=my_main, argv=[sys.argv[0]] + unparsed)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/tensorflow_core/python/platform/app.py", line 40, in run
    _run(main=main, argv=argv, flags_parser=_parse_flags_tolerate_undef)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/absl/app.py", line 299, in run
    _run_main(main, args)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/absl/app.py", line 250, in _run_main
    sys.exit(main(argv))
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/tensorflow_core/python/tools/freeze_graph.py", line 486, in <lambda>
    my_main = lambda unused_args: main(unused_args, flags)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/tensorflow_core/python/tools/freeze_graph.py", line 378, in main
    flags.saved_model_tags, checkpoint_version)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/tensorflow_core/python/tools/freeze_graph.py", line 361, in freeze_graph
    checkpoint_version=checkpoint_version)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/tensorflow_core/python/tools/freeze_graph.py", line 190, in freeze_graph_with_def_protos
    var_list=var_list, write_version=checkpoint_version)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/tensorflow_core/python/training/saver.py", line 828, in __init__
    self.build()
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/tensorflow_core/python/training/saver.py", line 840, in build
    self._build(self._filename, build_save=True, build_restore=True)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/tensorflow_core/python/training/saver.py", line 878, in _build
    build_restore=build_restore)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/tensorflow_core/python/training/saver.py", line 482, in _build_internal
    names_to_saveables)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/tensorflow_core/python/training/saving/saveable_object_util.py", line 343, in validate_and_slice_inputs
    for converted_saveable_object in saveable_objects_for_op(op, name):
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/tensorflow_core/python/training/saving/saveable_object_util.py", line 206, in saveable_objects_for_op
    variable, "", name)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/tensorflow_core/python/training/saving/saveable_object_util.py", line 83, in __init__
    self.handle_op = var.op.inputs[0]
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/tensorflow_core/python/framework/ops.py", line 2154, in __getitem__
    return self._inputs[i]
IndexError: list index out of range

Thanks for looking into this, Gerd

schelleg, 2021-05-05:
I just used the mnist notebook within a VAI 1.3.2 CPU docker (1.3.598) and builds the xmodels correctly.  Let me know if you're still seeing an issue.

schelleg, 2021-05-14:
closing due to inactivity ... please reopen as needed.

closed 2021-05-14 by schelleg