Xilinx/DPU-PYNQ #45: easy darknet support on host compiler

balintmaci, 2021-04-01:
The darknet models I've tried so far from [the model zoo](https://github.com/Xilinx/Vitis-AI/tree/master/models/AI-Model-Zoo) actually include the caffe prototxt and model files.
Just by changing `if [ $FRAMEWORK = 'cf' ]; then` to `if [ $FRAMEWORK = 'cf' ] || [ $FRAMEWORK = 'dk' ]; then` in both 2 places in the `compile.sh` I could use it to compile dk models from the zoo.
Granted I have not yet tested if the models work but at least the compilation ran without errors.

schelleg, 2021-05-05:
happy to accept a PR if that works for you ... we'll probably only test with cf/tf, but open to contributions.

schelleg, 2021-05-14:
closing due to inactivity ... please reopen as needed.

closed 2021-05-14 by schelleg