Xilinx/DPU-PYNQ #46: dpu.bit not found using Jupyter extension from VS Code remote

balintmaci, 2021-04-01:
I have a notebook using the DPU. When I run this notebook from the main Jupyter instance it works fine.
For easier development I'm trying to use VS Code over SSH remote with the Jupyter extension, which creates a separate Jupyter server and and displays the notebook within the IDE.
When I run the same notebook that way, the DpuOverlay cannot find the dpu bitstream.
Here's the exception message:
```
---------------------------------------------------------------------------
FileNotFoundError                         Traceback (most recent call last)
<ipython-input-5-d7286a9e2014> in <module>()
----> 1 overlay = DpuOverlay("dpu.bit")
      2 overlay.load_model("dpu_tf_inceptionv1.xmodel")

/home/xilinx/.local/lib/python3.6/site-packages/pynq_dpu/dpu.py in __init__(self, bitfile_name, dtbo, download, ignore_version, device)
     78             abs_bitfile_name = os.path.join(OVERLAY_PATH, bitfile_name)
     79         else:
---> 80             raise FileNotFoundError('Cannot find {}.'.format(bitfile_name))
     81         super().__init__(abs_bitfile_name,
     82                          dtbo=dtbo,

FileNotFoundError: Cannot find dpu.bit.
```
Any ideas what could be the problem?

schelleg, 2021-05-05:
I suppose it depends where the VSCode server is running from ... our notebooks are fragile in that they expect the dpu.bit in a known place... we probably won't be able to replicate your setup - perhaps try full path to dpu.bit or move it beforehand.

schelleg, 2021-05-14:
closing due to inactivity ... please reopen as needed.

closed 2021-05-14 by schelleg