Xilinx/DPU-PYNQ #51: Error with dpu.execute_async() using densebox_640_360.xmodel

majlucaaa, 2021-05-06:
Hi, I am currently working with DPU-PYNQ on ZCU104. I am working with a project of Face Tracking based on this project [avnet_face_tracking](https://github.com/Avnet/face_py_vart/blob/2020.2/avnet_face_tracking.py) but I have a problem  with the line "job_id = dpu.execute_async( inputData, outputData )" (inside the function "faces = dpu_face_detector.process(frame)"). When it reaches this part of the code, the kernel stops working, like if it was in an infinite loop, but it doesnt show any error. 
Can you tell me what the dpu.execute_async() function is expecting to do? I will send you my code:

`import sys
if '/usr/lib/python3.6/site-packages' not in sys.path:
    sys.path.append('/usr/lib/python3.6/site-packages')

import os
import subprocess
from ctypes import *
from typing import List
import pynq
import vart
import xir
from ctypes import *
from typing import List
import cv2
import numpy as np
import pathlib
import math
import threading
import time
import sys
import argparse

from imutils.video import FPS
from pyimagesearch.centroidtracker import CentroidTracker

from vitis_ai_vart.facedetect import FaceDetect
from vitis_ai_vart.utils import get_child_subgraph_dpu

from pynq_dpu import DpuOverlay
from matplotlib import pyplot as plt    # show image 

# construct the argument parse and parse the arguments
inputId = 0
detThreshold = 0.55
nmsThreshold = 0.35

densebox_xmodel = "densebox_640_360.xmodel"
overlay = DpuOverlay("dpu.bit")
print("[INFO] dpu overlay loaded")
overlay.load_model(densebox_xmodel)
# Initialize Vitis-AI/DPU based face detector
densebox_dpu = overlay.runner
dpu_face_detector = FaceDetect(densebox_dpu,detThreshold,nmsThreshold)       # Check face detect function
dpu_face_detector.start()

# Initialize our centroid tracker and frame dimensions
ct = CentroidTracker()
print("[INFO] Begin dpu process")

# Initialize the camera input
print("[INFO] starting camera input ...")
cam = cv2.VideoCapture(inputId)
cam.set(cv2.CAP_PROP_FRAME_WIDTH,640)
cam.set(cv2.CAP_PROP_FRAME_HEIGHT,480)
print("Capture device is open: " + str(cam.isOpened()))
if not (cam.isOpened()):
    print("[ERROR] Failed to open camera ", inputId )
    exit()
# START TRUE LOOP
# Capture image from camera
ret,frame = cam.read()
# Show image inside notebook using Matplotlib
plt.imshow(frame[:,:,[2,1,0]])
plt.show()
# Vitis-AI/DPU based face detector
faces = dpu_face_detector.process(frame)             #---------------THIS LINE IS NOT WORKING`

Inside this  dpu_face_detector.process(frame) function you can find this:
`def process(self,img):
    #print("[INFO] facefeature process")

    dpu = self.dpu
    #print("[INFO] facefeature runner=",dpu)

    inputChannels = self.inputChannels
    inputHeight = self.inputHeight
    inputWidth = self.inputWidth
    inputShape = self.inputShape
    output0Channels = self.output0Channels
    output0Height = self.output0Height
    output0Width = self.output0Width
    output0Size = self.output0Size
    output0Shape = self.output0Shape
    output1Channels = self.output1Channels
    output1Height = self.output1Height
    output1Width = self.output1Width
    output1Size = self.output1Size
    output1Shape = self.output1Shape

    imgHeight = img.shape[0]
    imgWidth  = img.shape[1]
    scale_h = imgHeight / inputHeight
    scale_w = imgWidth / inputWidth
    
    """ Image pre-processing """
    #print("[INFO] process - pre-processing - normalize ")
    # normalize
    img = img - 128.0
    #print("[INFO] process - pre-processing - resize ")
    # resize
    img = cv2.resize(img,(inputWidth,inputHeight))

    """ Prepare input/output buffers """
    #print("[INFO] process - prep input buffer ")
    inputData = []
    inputData.append(np.empty((inputShape),dtype=np.float32,order='C'))
    inputImage = inputData[0]
    inputImage[0,...] = img

    #print("[INFO] process - prep output buffer ")
    outputData = []
    outputData.append(np.empty((output0Shape),dtype=np.float32,order='C'))
    outputData.append(np.empty((output1Shape),dtype=np.float32,order='C'))

    """ Execute model on DPU """
    #print("[INFO] process - execute ")
    job_id = dpu.execute_async( inputData, outputData )                                                 #-----------THIS LINE IS NOT WORKING
    dpu.wait(job_id)

    """ Retrieve output results """    
    #print("[INFO] process - get outputs ")
    OutputData0 = outputData[0].reshape(1,output0Size)
    bboxes = np.reshape( OutputData0, (-1, 4) )
    #
    outputData1 = outputData[1].reshape(1,output1Size)
    scores = np.reshape( outputData1, (-1, 2))

    """ Get original face boxes """
    gy = np.arange(0,output0Height)
    gx = np.arange(0,output0Width)
    [x,y] = np.meshgrid(gx,gy)
    x = x.ravel()*4
    y = y.ravel()*4
    bboxes[:,0] = bboxes[:,0] + x
    bboxes[:,1] = bboxes[:,1] + y
    bboxes[:,2] = bboxes[:,2] + x
    bboxes[:,3] = bboxes[:,3] + y

    """ Run softmax """
    softmax = softmax_2( scores )

    """ Only keep faces for which prob is above detection threshold """
    prob = softmax[:,1] 
    keep_idx = prob.ravel() > self.detThreshold
    bboxes = bboxes[ keep_idx, : ]
    bboxes = np.array( bboxes, dtype=np.float32 )
    prob = prob[ keep_idx ]
	
    """ Perform Non-Maxima Suppression """
    face_indices = []
    if ( len(bboxes) > 0 ):
        face_indices = nms_boxes( bboxes, prob, self.nmsThreshold );

    faces = bboxes[face_indices]

    # extract bounding box for each face
    for i, face in enumerate(faces):
        xmin = max(face[0] * scale_w, 0 )
        ymin = max(face[1] * scale_h, 0 )
        xmax = min(face[2] * scale_w, imgWidth )
        ymax = min(face[3] * scale_h, imgHeight )
        faces[i] = ( int(xmin),int(ymin),int(xmax),int(ymax) )

    return faces`

schelleg, 2021-05-10:
did you recompile the xmodel for the ZCU104's DPU?

majlucaaa, 2021-05-11:
what do you mean?

schelleg, 2021-05-14:
Its part of Vitis-AI, I see it in their repo/docs:
https://www.xilinx.com/support/documentation/sw_manuals/vitis_ai/1_3/ug1414-vitis-ai.pdf 

Not sure if you can find more details there..

skalade, 2022-02-14:
Hopefully this has been resolved by now... Closing the issue due to lack of activity.

closed 2022-02-14 by skalade

Mushegh097, 2023-06-16:
I am having issue with the dpu.execute_async function on ZCU104 as well:

"
WARNING: Logging before InitGoogleLogging() is written to STDERR
W0616 21:44:00.953873  1599 xrt_cu.cpp:188] cu timeout! device_core_idx 0  handle=0xaaaad8697d70 ENV_PARAM(XLNX_DPU_TIMEOUT) 10000 state 1 ERT_CMD_STATE_COMPLETED 4 ms 10010  bo=1 is_done 0
I0616 21:44:00.953997  1599 xrt_cu.cpp:99] Total: 10010290us    ToDriver: 18446740573875003us   ToCU: 0us       Complete: 0us      Done: 3509844838us
F0616 21:44:00.954035  1599 dpu_control_xrt_edge.cpp:186] dpu timeout! core_idx = 0
 LSTART 0  LEND 0  CSTART 0  CEND 0  SSTART 1023  SEND 0  MSTART 1023  MEND 0  CYCLE_L 0  CYCLE_H 0
*** Check failure stack trace: ***
Aborted (core dumped)
"

Can sm1 help?