Xilinx/DPU-PYNQ #53: compatibility with pynq v2.5

daivikbhatia, 2021-05-11:
Hi for some reason I have to use PYNQ 2.5 image, Will this upgrade work for version 2.5 too?

daivikbhatia, 2021-05-18:
Hi thankyou for the reply, I tested the upgrade on version 2.5 but ran into following error:
[image]
can you please help me with it?

schelleg, 2021-05-18:
If you look at v1.2.0, you'll get a hint at the difference between a PYNQ v2.5 upgrade and a PYNQ v2.6 upgrade.

See this version of xrt upgrade here: https://github.com/Xilinx/DPU-PYNQ/blob/v1.1.2/upgrade/xrt/install.sh#L14

daivikbhatia, 2021-05-21:
Hi, thanks that helped. I'm installing the whole v1.1.2 on my pynq v2.5. The upgradation ran successfully but when I tried to import pynq-dpu, then I ran into following error:
[image]
[image]

Maybe it's because v1.1.2 installed everything on pynq version 2.5.1 but when I installed pynq-dpu with pip3 install pynq-dpu , it replaced pynq version 2.5.1 with version 2.6.0. If that's the case can you please tell me the correct pynq-dpu version for v1.1.2?
or any other way to degug this error?

daivikbhatia, 2021-05-30:
Hi, Thank-you so much for your help. I was successful in running v1.1.2 on pynq v2.5. 
Do you have any idea how to run yolo-v3 ipynb notebook for multiple images. it's running fine on 1 image but kernal crashes when I loop it for multiple images. Any idea how can I resolve that or make code to run for multiple images?

skalade, 2022-02-14:
Closing the issue due to lack of activity. Yolo notebooks that were based on DNNDK are no longer supported by versions of Vitis AI >1.3, best place to look how to run them will be in [older documentation](https://www.xilinx.com/support/documentation/sw_manuals/vitis_ai/1_2/ug1414-vitis-ai.pdf).

closed 2022-02-14 by skalade