Xilinx/DPU-PYNQ #54: Makefile:7: recipe for target 'check_env' failed

chumingqian, 2021-05-14:
Hi, all:
   Is there anyone who met the following problem:
 

root@pynq:/home/xilinx# git clone --recursive --shallow-submodules https://github.com/Xilinx/DPU-PYNQ
Cloning into 'DPU-PYNQ'...
remote: Enumerating objects: 325, done.
remote: Counting objects: 100% (40/40), done.
remote: Compressing objects: 100% (32/32), done.
remote: Total 325 (delta 7), reused 26 (delta 5), pack-reused 285
Receiving objects: 100% (325/325), 3.55 MiB | 1.28 MiB/s, done.
Resolving deltas: 100% (143/143), done.
Submodule 'vitis-ai-git' (https://github.com/Xilinx/Vitis-AI.git) registered for path 'vitis-ai-git'
Cloning into '/home/xilinx/DPU-PYNQ/vitis-ai-git'...
remote: Enumerating objects: 26536, done.
remote: Counting objects: 100% (26536/26536), done.
remote: Compressing objects: 100% (18395/18395), done.
remote: Total 26536 (delta 7271), reused 25480 (delta 7055), pack-reused 0
Receiving objects: 100% (26536/26536), 563.72 MiB | 2.06 MiB/s, done.
Resolving deltas: 100% (7271/7271), done.
fatal: unable to access 'https://github.com/Xilinx/Vitis-AI.git/': Empty reply from server
Unable to fetch in submodule path 'vitis-ai-git'
root@pynq:/home/xilinx# cd DPU-PYNQ/upgrade
root@pynq:/home/xilinx/DPU-PYNQ/upgrade# make
bash ./check_env.sh
Error: please make sure you have valid Internet connection.
Makefile:7: recipe for target 'check_env' failed
make: *** [check_env] Error 1

schelleg, 2021-05-14:
Looks like internet connection went down midway through the clone?  I tried the clone and working for me today.

chumingqian, 2021-05-16:
No, I can connect to the network, I just can not make Makefile, it referred to the bash. / check _env.sh.

schelleg, 2021-05-16:
I'm just looking at the error message and I see:
```
fatal: unable to access 'https://github.com/Xilinx/Vitis-AI.git/': Empty reply from server
Unable to fetch in submodule path 'vitis-ai-git'
```

Can you retry the clone and see if those messages go away ?  Also that checkenv script is right there in the folder, you can see what it is trying to do.

https://github.com/Xilinx/DPU-PYNQ/blob/master/upgrade/check_env.sh#L19

chumingqian, 2021-05-17:
Appreciated, it worked!

closed 2021-05-17 by chumingqian