Xilinx/DPU-PYNQ #56: PYNQ-derivative-overlays.git missing for DPU-PYNQ v1.1.2

daivikbhatia, 2021-07-12:
Hi,
I'm using dpu-pynq v1.1.2 for pynq 2.5 image. I want to make cutom DPU as suggested here https://github.com/Xilinx/DPU-PYNQ/tree/v1.1.2/boards
but unfortunately when i run the make command for Ultra96 v2, it tries to clone this repository https://github.com/yunqu/PYNQ-derivative-overlays.git and this has been moved. 
Can you please give me new path of this repository for pynq-dpu v1.1.2 ?

schelleg, 2021-07-12:
hi @daivikbhatia - can you upgrade to v1.3.2?  That repo was taken down as you saw.  

Otherwise, you can try the steps in the current Makefile which contain the platform building steps and apply it to your custom design.

luke-avionics, 2021-07-21:
hi @daivikbhatia 

do you run the make command in 1.3.2 and keep the image with dpu-pynq 1.1.2 so it is compatible with pynq 2.5? 

Or do you make any modifications to the makefile in 1.1.2? 

Thank you

daivikbhatia, 2021-07-29:
Hi @luke-avionics 
I tried it with running make command in 1.3.2 and keep the image with dpu-pynq 1.1.2 so it is compatible with pynq 2.5 but it failed.
I have not tried modifications to the makefile in 1.1.2 yet.

daivikbhatia, 2021-07-29:
Hi @schelleg 
Thankyou so much for the reply. Is there anyway to run PYNQ 2.5 with pynq dpu v1.3.2? If there is a way, it will solve a lot of problems for me.

schelleg, 2021-07-30:
you can try mish-mashing the upgrade scripts together, they are somewhat hardcoded to specific Linux kernel versions which means hardcoded paths ... have you tried it out?  Otherwise, hard for us to support untested combinations of VitisAI and PYNQ

skalade, 2022-02-14:
Closing the issue due to lack of activity.

closed 2022-02-14 by skalade