Xilinx/DPU-PYNQ #57: No module named 'vart'

wanghao16020510036, 2021-07-12:
When I try run "dpu_mnist_classifier.ipynb" in "pynq_dpu/edge/notebooks", I got the following error message:
---
ModuleNotFoundError                       Traceback (most recent call last)
<ipython-input-1-73f05173f865> in <module>()
----> 1 from pynq_dpu import DpuOverlay
      2 overlay = DpuOverlay("dpu.bit")
      3 overlay.load_model("dpu_mnist_classifier.xmodel")

/usr/local/lib/python3.6/dist-packages/pynq_dpu-1.3.2-py3.6-linux-aarch64.egg/pynq_dpu/__init__.py in <module>()
     14 
     15 
---> 16 from .dpu import DpuOverlay
     17 
     18 

/usr/local/lib/python3.6/dist-packages/pynq_dpu-1.3.2-py3.6-linux-aarch64.egg/pynq_dpu/dpu.py in <module>()
     24 from typing import List
     25 import pynq
---> 26 import vart
     27 import xir
     28 

ModuleNotFoundError: No module named 'vart'
---
I know this is because "vart" is not installed and I need install it. But after I operated followed [Vitis AI UG](https://china.xilinx.com/html_docs/vitis_ai/1_3/installation.html#qoi1603967292432) and [readme](https://github.com/Xilinx/Vitis-AI/tree/master/setup/mpsoc/VART#step1-setup-cross-compiler), this problem is still not solved. Very eager for your guidance!

schelleg, 2021-07-12:
VART is installed during dpu-pynq's install, see the setup.py line below:

https://github.com/Xilinx/DPU-PYNQ/blob/master/setup.py#L104 

Did your dpu-pynq install complete successfully?

wanghao16020510036, 2021-07-13:
> VART is installed during dpu-pynq's install, see the setup.py line below:
> 
> https://github.com/Xilinx/DPU-PYNQ/blob/master/setup.py#L104
> 
> Did your dpu-pynq install complete successfully?

Thank you very much for your reply! I reinstalled pynq-dpu. This problem has been resolved.

closed 2021-07-29 by schelleg