Xilinx/DPU-PYNQ #64: DPU example: MNIST Classifier

Afef00, 2021-08-13:
Is the model used in this example came from the trained model on this notebook [https://github.com/Xilinx/DPU-PYNQ/blob/master/host/train_mnist_model.ipynb](url)?
I wanted to know the architecture of the used model ("dpu_mnist_classifier.xmodel").

schelleg, 2021-08-13:
yes - dpu_mnist_classifier.xmodel is generated from the  train_mnist_model.ipynb

Afef00, 2021-08-13:
OK, thank you!

Afef00, 2021-08-13:
One more question please.

After successfully running the MNIST classifier example on the ZCU104 board, I wanted to store the final extracted features of all the test samples to be used as training examples for the SVM classifier using : result. append (data) as shown in the code below. However, I found out that only the features of one test sample are stored and I don't know how to do this. Also, the results array should be in the shape of (nsamples, nfeatures) to be fed to SVM classifier. I tried results = np.array ((total, outputSize), dtype=float) for the declaration of the results array but it didn't work.

Any suggestions please?
`from time import time
import sklearn
import numpy as np
import mnist
import cv2
from sklearn import svm
import matplotlib.pyplot as plt
%matplotlib inline
from six.moves import urllib
opener = urllib.request.build_opener()
opener.addheaders = [('User-agent', 'Mozilla/5.0')]
urllib.request.install_opener(opener)
from pynq_dpu import DpuOverlay
overlay = DpuOverlay("dpu.bit")
overlay.load_model("dpu_mnist_classifier.xmodel")
raw_data = mnist.test_images()
normalized_data = np.asarray(raw_data/255, dtype=np.float32)
test_data = np.expand_dims(normalized_data, axis=3)
test_label = mnist.test_labels()
dpu = overlay.runner

inputTensors = dpu.get_input_tensors()
outputTensors = dpu.get_output_tensors()

shapeIn = tuple(inputTensors[0].dims)
shapeOut = tuple(outputTensors[0].dims)
outputSize = int(outputTensors[0].get_data_size() / shapeIn[0])
softmax = np.empty(outputSize)
output_data = [np.empty(shapeOut, dtype=np.float32, order="C")]
input_data = [np.empty(shapeIn, dtype=np.float32, order="C")]
image = input_data[0]
def calculate_softmax(data):
result=[]
result.append(data)
return result
total = test_data.shape[0]
for i in range(total):
    image[0,...] = test_data[i]
    job_id = dpu.execute_async(input_data, output_data)
    dpu.wait(job_id)
    temp = [j.reshape(1, outputSize) for j in output_data]
    softmax = calculate_softmax(temp[0][0])
    array2 = np.array(softmax)
    print("results",(array2))`

closed 2021-08-16 by schelleg