Xilinx/DPU-PYNQ #72: SSD Object Detector Dual output

bomerzz, 2022-01-17:
Hello,

I've been trying to run the [tf2_ssdincepion_v2](https://github.com/Xilinx/Vitis-AI/tree/master/models/AI-Model-Zoo/model-list/tf_ssdinceptionv2_coco_300_300_9.62G_1.4) model using dpu-pynq using a modified version of the dpu_tf_inceptionv1.ipynb file.

By using the defaults of  `shapeOut = tuple(outputTensors[0].dims)` I was able to get an output array of shape (1,1917,91) which I assume is the confidence score for each box.

If I access the second output tensor `shapeOut2 = tuple(outputTensors[1].dims)
` it will give me a shape of (1,1917,4) which I assume will be the box locations.

However while trying to run with the second set of outputensors, the following error will be thrown:

```
 job_id = dpu.execute_async(input_data, output_data2)
double free or corruption (out)
Aborted (core dumped)
```

I am not sure how to proceed with doing the detection as from the graph generated using  analyze_subgraphs.sh the subgraph has 2 outputs that require CPU processing to obtain the final result.

Any help will be appreciated.  Thank you!

[Graph Image](https://imgur.com/NRSFU5E.jpg
)

skalade, 2022-01-17:
Hi there,

I think this is a limitation DpuOverlay, looking at [dpu.py](https://github.com/Xilinx/DPU-PYNQ/blob/master/pynq_dpu/dpu.py#L172) the first and only subgraph is used to create a runner object. We might support subgraphs in the future, but for now you would have to use vart directly and run separate instances [like here](https://github.com/Xilinx/Vitis-AI-Tutorials/blob/master/Feature_Tutorials/pytorch-subgraphs/files/application/main_subgraphs.py#L307). 

Thanks
Shawn

bomerzz, 2022-01-18:
Hi @skalade thanks for the reply.
I've tried the above mentioned method but it will still throw the double free error when attempting to use an output buffer of size of the second output (1,1917,4).  Any ideas how I might got about solving this?

skalade, 2022-01-18:
Hi, since this does not seem like a DPU-PYNQ bug I'd recommend posting this on the [pynq discuss](https://discuss.pynq.io/) forum or maybe as a more general vart question to the xilinx forums / Vitis AI github issues. 

I've not really worked with models like ssd before so can't give too much advice. But parsing your model looks like the 4th CPU subgraph corresponds to the box encodings that you can see on one of your outputs in that graph image you provided. So maybe you could grab those somehow, there should be some examples in the Vitis AI library.

[image]

Hope this helps a bit...

I'm going to close this issue because this isn't a core DPU-PYNQ problem. If you still have issues I encourage you to post on one of the forums!

Thanks
Shawn

closed 2022-01-18 by skalade

bomerzz, 2022-01-19:
Hi @skalade thanks for the help! I'll give the forum a shot.