Xilinx/DPU-PYNQ #73: About deploying to the board

CheungBH, 2022-01-17:
Hello. Thanks for your work. I am a newcomer of DPU and Vitis-AI, and I feel confused about the procedure for deployment.

You mentioned that "we first need to make sure Vitis and XRT have been installed and their settings are sourced properly."

**_source <vitis-install-path>/Vitis/2021.1/settings64.sh
source <xrt-install-path>/xilinx/xrt/setup.sh_**

I wonder are there any references related to them? I could not find the corresponding repo or files. And how to check whether Vitis and XRT are installed?

Thank you.

skalade, 2022-01-17:
Hi there,

If you have a supported board like a ZCU104 or Ultra96, you do not have to rebuild the designs. Simply download the [latest PYNQ image for your board](http://www.pynq.io/board.html), and once you have that set up you can pip install the dpu-pynq package, which will have the bitstreams and models pre-compiled.

For rebuilding the hardware designs you will need Xilinx development tools (these are not part of this repository), such as Vitis, which you can read more about and download [here](https://www.xilinx.com/support/download/index.html/content/xilinx/en/downloadNav/vivado-design-tools/2021-1.html). XRT is a runtime library which a lot of packages in PYNQ and Vitis AI use, you can read more about it [here](https://www.xilinx.com/products/design-tools/vitis/xrt.html). Once you install them on your machine, you usually have to source some bash scripts (such as `settings64.sh`) from their install locations in order to be able to launch them and build your designs.

Thanks
Shawn

CheungBH, 2022-01-19:
Thank you for your reply. Actually, I am currently installing XRT. However, I couldn't find the target file in "/xilinx/xrt/setup.sh". Instead, the only setup.sh is found in "/XRT/src/runtime_src/tools/scripts/setup.sh". Is it the correct one?

CheungBH, 2022-01-19:
I obtained the hardware design file ".bit, .hwh, and .xclbin". Also, I got the AI model ".xmodel".
And I think the next step is to run build the model into the hardware. Are there any references or tutorials about it? I am not very clear about how to integrate them.

skalade, 2022-01-19:
Well that's what the example notebooks are for https://github.com/Xilinx/DPU-PYNQ/tree/master/pynq_dpu/notebooks

Other than that you can find plenty of examples in [Vitis AI tutorials](https://github.com/Xilinx/Vitis-AI-Tutorials) and [Vitis AI github repo](https://github.com/Xilinx/Vitis-AI).

CheungBH, 2022-01-20:
Thank you. I am trying to install pynq-dpu.
The command of "pip3 install pynq-dpu --no-build-isolation" is executed successfully. However, when I want to use pynq_dpu library, an error occurs showing in the figure. 
Are there anything missing?
[image]

skalade, 2022-01-20:
Looks like you're using a pynq v2.6 image. The current release of dpu-pynq is supported for v2.7. You can download the latest one from [here](http://www.pynq.io/board.html) if you are using a supported board.

CheungBH, 2022-01-21:
Unfortunately, I cannot update the board version since there are many existing projects belonging to other people...
I wonder whether I can use an older version of dpu which is compatible with v2.6? And which version will be suitable?

skalade, 2022-02-14:
I will close this out as this seems to be an old example not running on a newer image which might be incompatible. Please do reply if you have found a solution.

closed 2022-02-14 by skalade