Xilinx/DPU-PYNQ #76: How to get the output_data using VART API in PYNQ-DPU

Afef00, 2022-01-28:
Hello,
I was trying to run a custom CNN model using PYNQ-DPU on ZCU104 board following [this example](https://github.com/Xilinx/DPU-PYNQ/blob/master/pynq_dpu/notebooks/dpu_mnist_classifier.ipynb). And I was trying to retrieve the data from the layer before Softmax. 
As far as I know, the output data here is the output data of the layer before Softmax , which is the dense layer in the attached example of a model architecture. The shape of the output data will be (1,128).  Am I correct ???
`    job_id = dpu.execute_async(input_data, output_data)
    dpu.wait(job_id)
    temp = [j.reshape(1, outputSize) for j in output_data]
    softmax = calculate_softmax(temp[0][0])`
[image]
When I tried to reshape the output data to be (1,128), there was  NaN in the data.

So I tried this:
After the training of the full model using tensorflow, I run vai_q_tensorflow to obtain a .pb quantized model. I truncate this model  to delete softmax layer. The truncated model (conv+pool+dense1) is compiled using 
```
!vai_c_tensorflow \
    --frozen_pb custom_dense1.pb \
    --arch /opt/vitis_ai/compiler/arch/DPUCZDX8G/ZCU104/arch.json \
    --output_dir . \
    --net_name mnist_dense1_modif
```
However I got the following error
**************************************************
* VITIS_AI Compilation - Xilinx Inc.
**************************************************
[INFO] Namespace(inputs_shape=None, layout='NHWC', model_files=['custom_dense1.pb'], model_type='tensorflow', out_filename='./mnist_dense1_modif_org.xmodel', proto=None)
[INFO] tensorflow model: custom_dense1.pb
[INFO] parse raw model     :  8%| | 1/12 [00:00<00:00, 3675.99it/s]             
Traceback (most recent call last):
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/bin/xnnc-run", line 33, in <module>
    sys.exit(load_entry_point('xnnc==1.3.0', 'console_scripts', 'xnnc-run')())
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/xnnc/__main__.py", line 194, in main
    normal_run(args)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/xnnc/__main__.py", line 178, in normal_run
    in_shapes=in_shapes if len(in_shapes) > 0 else None,
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/xnnc/xconverter.py", line 131, in run
    xmodel = CORE.make_xmodel(model_files, model_type, _layout, in_shapes)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/xnnc/core.py", line 104, in make_xmodel
    model_files, layout, in_shapes=in_shapes, model_type=model_t
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/xnnc/translator/tensorflow_translator.py", line 97, in to_xmodel
    model_name, raw_nodes, layout, in_shapes, model_fmt, model_type
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/xnnc/translator/tensorflow_translator.py", line 161, in create_xmodel
    xmodel = cls.__create_xmodel_from_tf1(name, layers, layout, in_shapes)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/xnnc/translator/tensorflow_translator.py", line 243, in __create_xmodel_from_tf1
    xmodel_name, layout, layers, const_layer_dict, super_const_dict, in_shapes
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/xnnc/translator/tensorflow_translator.py", line 1847, in __generate_xmodel
    ), f"[ERROR] TF Conv2d requires two inputs: actual: {bottom}."
AssertionError: [ERROR] TF Conv2d requires two inputs: actual: ['input_data'].

Any suggestions how to solve this problem please?
Thanks.

skalade, 2022-01-28:
The model in the example already outputs the final fully connected layer output values before softmax and its shape is (1, 10). Softmax is calculated as a separate python function.

If you are having issues training custom models on the Vitis AI docker then [Vitis AI](https://github.com/Xilinx/Vitis-AI) would be a better place to find support. I would also investigate the [Vitis AI Tutorials](https://github.com/Xilinx/Vitis-AI-Tutorials) as there are plenty of training examples of more complex architectures.

Thanks
Shawn

Afef00, 2022-01-29:
Hello,
Thanks for the clarification!  
Actually, I was trying to  build a hybrid CNN -SVM model, where I use the output values of the final fully connected layer as inputs for SVM classifier. 
The problem is that in the software validation of this approach the output shape of this layer was (1,128).

skalade, 2022-02-14:
Closing this out as it has been a few weeks without activity and it seems more of a general ML/tensorflow issue, than to do with DPU-PYNQ.

closed 2022-02-14 by skalade