Xilinx/DPU-PYNQ #79: pip3 does not deploy customized DPU

haipnh, 2022-03-19:
Hi,

After successfully building a single core DPU for ZCU104, I manually copied DPU-PYNQ directory to my board I ran this command:

```shell
pip3 install /home/xilinx/DPU-PYNQ --no-build-isolation
```

I opened a notebook, created a overlay object to load the dpu.bit, and investigate its properties by `overlay?` command. It showed:

```
Type:            DpuOverlay
String form:     <pynq_dpu.dpu.DpuOverlay object at 0xffff659435b0>
File:            /usr/local/share/pynq-venv/lib/python3.8/site-packages/pynq_dpu/dpu.py
Docstring:      
Default documentation for overlay /usr/local/share/pynq-venv/lib/python3.8/site-packages/pynq_dpu/dpu.bit. The following
attributes are available on this overlay:

IP Blocks
----------
axi_vip_2            : pynq.overlay.DefaultIP
DPUCZDX8G_1          : pynq.overlay.DefaultIP
DPUCZDX8G_2          : pynq.overlay.DefaultIP
axi_vip_1            : pynq.overlay.DefaultIP
ps_e_0               : pynq.overlay.DefaultIP
```

So I went to the Jupyter terminal and manually copied the required files:
```shell
cd /home/xilinx/DPU-PYNQ/boards/ZCU104
cp dpu.bit dpu.hwh ../vitis_platform/dpu/dpu.tcl binary_container_1/dpu.xclbin /usr/local/share/pynq-venv/lib/python3.8/site-packages/pynq_dpu
```

I restarted the resnet50 again, ran  `overlay?` command again and it showed:
```
Type:            DpuOverlay
String form:     <pynq_dpu.dpu.DpuOverlay object at 0xffff659434f0>
File:            /usr/local/share/pynq-venv/lib/python3.8/site-packages/pynq_dpu/dpu.py
Docstring:      
Default documentation for overlay /usr/local/share/pynq-venv/lib/python3.8/site-packages/pynq_dpu/dpu.bit. The following
attributes are available on this overlay:

IP Blocks
----------
axi_vip_2            : pynq.overlay.DefaultIP
DPUCZDX8G_1          : pynq.overlay.DefaultIP
axi_vip_1            : pynq.overlay.DefaultIP
ps_e_0               : pynq.overlay.DefaultIP
```

My question is how can we support to looking for the local bitstream files before getting online prebuilt ones?

Thank you in advance.

skalade, 2022-03-20:
Hi there,

That's just the [order of priority](https://github.com/Xilinx/DPU-PYNQ/blob/master/pynq_dpu/dpu.py#L80) of the dpu.py module. You could try to pass it an absolute path to where your custom bitstream resides and that should work without having to copy files over. That is a good point, in future releases we could consider swapping the priority order to look for a local bitstream first...

Thanks
Shawn

haipnh, 2022-03-21:
Hi @skalade ,

Thank you for replying.

I mean looking for local bitstream first when use pip3 command, not when use dpu.py module.

haipnh, 2022-03-21:
I close this issue and look forward to next releases.

closed 2022-03-21 by haipnh