Xilinx/DPU-PYNQ #80: Deployment of Softmax in hardware

Afef00, 2022-03-19:
Hi,
I Know that Softmax operations was done on CPU and the other operations by DPU kernel.
I was wondering is it possible to re-configure the DPU hardware design so all the operations of softmax and DPU are done in hardware?
In that case how to modify the inference file ?
Regards

skalade, 2022-03-20:
Hi there,

The [Vitis AI DPU-TRD](https://github.com/Xilinx/Vitis-AI/tree/1.4/dsa/DPU-TRD/prj/Vitis) project shows how to build the softmax IP as well as a DPU design. I would consult the materials on that repo. And I'm sure there should be a few examples using it on [Vitis AI tutorials](https://github.com/Xilinx/Vitis-AI-Tutorials). There are some relevant issues on the Vitis AI repo as well https://github.com/Xilinx/Vitis-AI/issues/131 and https://github.com/Xilinx/Vitis-AI/issues/221. 

This isn't really a DPU-PYNQ issue, so I'll close it for now. But if you have similar concerns I'd encourage you to make a forum post on the Xilinx support forums, or if it is more to do with dpu-pynq, on the pynq discuss forum.

Thanks
Shawn

closed 2022-03-20 by skalade

Afef00, 2022-03-21:
Thank your for the help!