Xilinx/DPU-PYNQ #84: Hardware utilization of zcu104

HenningCode, 2022-04-28:
Hello Guys,

I rebuild the Hardware for myself with a few small changes. These changes can be seen here:https://github.com/HenningCode/DPU-PYNQ 

Now I wanted to look at the hardware utilization and the utilization is surprisingly low and I dont understand why. The generated bitstream and everything works but it seems like the DPU is missing from the BD?  Is there a way to find the "real" hardware utilization?
[image]

skalade, 2022-04-28:
Hi there,

I assume you are looking at the output of the dpu.tcl script which generates a Vitis platform. The DPU is added in the Vitis flow with V++, that generates a separate Vivado project which you can look at in e.g. boards/ZCU104/binary_container_1/link/vivado/vpl/prj/

There's some similar Xilinx forum posts on this like the one [here](https://support.xilinx.com/s/question/0D52E00006hpSOmSAM/resource-utilization-of-a-model-on-zcu104-in-vitis-ai-13?language=en_US).

Thanks
Shawn

skalade, 2023-05-09:
Hopefully the answer has been sufficient and due to lack of updates I'm assuming issue has been resolved.

closed 2023-05-09 by skalade