Xilinx/DPU-PYNQ #92: ultra96 board resetting in different dpu_conf.vh and prj_config configurations?

anirudhitagi, 2022-09-24:
Greetings,

I have been using the [DPU-PYNQ](https://github.com/Xilinx/DPU-PYNQ) Version 1.4 to deploy a DPU on the Ultra96 board and run inference on it.

While I have been able to modify the dpu_conf.vh for some configurations and able to run inference with them, I attempted to modify the prj_config file to connect 300 MHz and 600 MHz clocks (replacing the 200 and 400 MHz clock).

For reference here is the changed section of the prj_config:
```
# clock availability
#  id=0 150MHz
#  id=1 300MHz
#  id=2  75MHz
#  id=3 100MHz
#  id=4 200MHz
#  id=5 400MHz
#  id=6 600MHz

#freqHz=300000000:DPUCZDX8G_1.aclk
#freqHz=600000000:DPUCZDX8G_1.ap_clk_2
#freqHz=300000000:DPUCZDX8G_2.aclk
#freqHz=600000000:DPUCZDX8G_2.ap_clk_2
#using 300/600 MHz
id=1:DPUCZDX8G_1.aclk
id=6:DPUCZDX8G_1.ap_clk_2
# 200Mhz/400MHz ---- previous/original configuration
#id=4:DPUCZDX8G_1.aclk
#id=5:DPUCZDX8G_1.ap_clk_2
```

I did the make=Ultra96 and that ran successfully and the dpu.bit, dup.hwh & dpu.xclbin files were also generated.

```
The bit files were generate successfully without any timing issues:
---------------------------------------------------------------------------------
| Design Timing Summary
| ---------------------
---------------------------------------------------------------------------------

    WNS(ns)      TNS(ns)  TNS Failing Endpoints  TNS Total Endpoints      WHS(ns)
    -------      -------  ---------------------  -------------------      -------
      0.017        0.000                      0               217598        0.010

All user specified timing constraints are met.

------------------------------------------------------------------------------------------------
| Clock Summary
| -------------
------------------------------------------------------------------------------------------------

Clock                         Waveform(ns)         Period(ns)      Frequency(MHz)
-----                         ------------         ----------      --------------
clk_pl_0                      {0.000 5.000}        10.000          100.000         
clk_pl_1                      {0.000 5.000}        10.000          100.000         
clk_pl_2                      {0.000 5.000}        10.000          100.000         
clk_pl_3                      {0.000 5.000}        10.000          100.000         
dpu_i/clk_wiz_0/inst/clk_in1  {0.000 5.000}        10.000          100.000         
  clk_out2_dpu_clk_wiz_0_0    {0.000 1.667}        3.333           300.000         
  clk_out3_dpu_clk_wiz_0_0    {0.000 6.667}        13.333          75.000          
  clk_out7_dpu_clk_wiz_0_0    {0.000 0.833}        1.667           600.000   
```      

I opened the vivado model and I can see that the clock connections are done as expected. [ 300 MHz connected to DPUCZDX8G_1.aclk and 600 MHz connected to DPUCZDX8G_1.ap_clk_2]

I am able to successfully program the board as well with `DpuOverlay` [ the D1 BLUE LED turns ON  + D5 RED LED goes OFF]
Here is the output of the overlay:

```
IP Blocks
----------
axi_vip_2            : pynq.overlay.DefaultIP
DPUCZDX8G_1          : pynq.overlay.DefaultIP
axi_vip_1            : pynq.overlay.DefaultIP
ps_e_0               : pynq.overlay.DefaultIP

```

But when I begin the inference with:
```
job_id = dpu.execute_async(input_data, output_data)
dpu.wait(job_id)
```
**The Ultra96-v2 board resets I believe. I can see that the D5 RED LEDs turns on indicating that it is not programmed and D3,D4,D6,D7 GREEN LEDs turn on and then off after a few seconds. during which the jupyter notebook is unresponsive.**

Can anyone please help me understand why this is happening? Bit file generated successfully without timing issues, and there is no problem with the xmodel or the way I am using dpu.execute_async(input_data, output_data), because this is working without any issue for the default prj_config file [200/400 MHz]. 

Any advice would be appreciated.

Thanks & Regards,
-AI

Daniel-bboy, 2022-10-10:
hello, I have met the same problem! I can also get the bit, hwh, xclbin file successfully. But, when i run: 
 job_id = dpu.execute_async(input_data, output_data)
dpu.wait(job_id)
 the kernel will die.

So have you solved the problem? Thx for your reply!

anirudhitagi, 2022-11-16:
Hi @focalplane ,
Yes If I run with the 200/400 MHz clock frequency it works.
But I ordered the 12v 4A power supply cable and I am trying with that, but then also the same error happens

focalplane, 2022-11-20:
Hi @anirudhitagi,  The external power supply is not really the issue.  It is the PMIC regulators on the U96v2 board that limit the power going to the ZU3 part.  That is why the DPU PYNQ examples for U96 have been compiled to run at the slower frequencies.

Kind regards

anirudhitagi, 2022-11-20:
@focalplane , Thank you for your response.
May I ask if any clock gating measures may help in this case?

skalade, 2023-11-29:
Closing this due to lack of activity. It looks like lowering the clock frequencies is what resolves it. Please re-open if there's something to be addressed on the dpu-pynq package side.