Xilinx/DPU-PYNQ #96: Error to compile a model _ VITIS_AI Compilation

Afef00, 2022-11-20:
I ran Vitis AI CPU docker container from host using ./docker_run.sh xilinx/vitis-ai-cpu:1.3.411. After successfully training my custom model using the following code:
```
import os
import numpy as np
import tensorflow as tf
from sklearn.metrics import confusion_matrix
image_ordering_dim="tf"
def preprocess(X_train, X_test):
    """
    Convert from float64 to float32 and normalize normalize to decibels
    relative to full scale (dBFS) for the 4 sec clip.
    """
    X_train = X_train.astype('float32')
    X_test = X_test.astype('float32')

    X_train = np.array([(X - X.min()) / (X.max() - X.min()) for X in X_train])
    X_test = np.array([(X - X.min()) / (X.max() - X.min()) for X in X_test])
    return X_train, X_test
def prep_train_test(X_train, y_train, X_test, y_test, nb_classes):
    """
    Prep samples ands labels for Keras input by noramalzing and converting
    labels to a categorical representation.
    """
    print('Train on {} samples, validate on {}'.format(X_train.shape[0],
                                                       X_test.shape[0]))

    # normalize to dBfS
    X_train, X_test = preprocess(X_train, X_test)

    # Convert class vectors to binary class matrices
    Y_train = tf.keras.utils.to_categorical(y_train, nb_classes)
    Y_test = tf.keras.utils.to_categorical(y_test, nb_classes)

    return X_train, X_test, Y_train, Y_test
def keras_img_prep(X_train, X_test, img_dep, img_rows, img_cols):

    X_train = X_train.reshape(X_train.shape[0], img_rows, img_cols, 1)
    X_test = X_test.reshape(X_test.shape[0], img_rows, img_cols, 1)
    input_shape = (img_rows, img_cols, 1)

    return X_train, X_test, input_shape

def cnn(X_train, y_train, X_test, y_test, batch_size,
        nb_classes, epochs, input_shape):
    """
  
    """

  inputs = tf.keras.Input(shape=input_shape)
  x = tf.keras.layers.Conv2D(128, (3,3), activation='relu', input_shape=(input_shape))(inputs)
  x = tf.keras.layers.MaxPooling2D((4,3))(x)
  x = tf.keras.layers.Conv2D(32, (1,3), activation='relu')(x)
  x = tf.keras.layers.MaxPooling2D((1,3))(x)
  x = tf.keras.layers.Flatten()(x)
  x = tf.keras.layers.Dense(512, activation='relu')(x)
  x = tf.keras.layers.Dense(512, activation='relu')(x)
  outputs = tf.keras.layers.Dense(2, activation='softmax')(x)

  model = tf.keras.Model(inputs=inputs, outputs=outputs, name='classifier')
    
    model.compile(loss='categorical_crossentropy',
                  optimizer='adadelta',
                  metrics=['accuracy'])

    history = model.fit(X_train, y_train, batch_size=batch_size, epochs=epochs,
                        verbose=1, validation_data=(X_test, y_test))

    # Evaluate accuracy on test and train sets
    score_train = model.evaluate(X_train, y_train, verbose=0)
    print('Train accuracy:', score_train[1])
    score_test = model.evaluate(X_test, y_test, verbose=0)
    print('Test accuracy:', score_test[1])
    model.summary()

    return model, history

    X_train = np.load('train_samples.npz')
    y_train = np.load('train_labels.npz')
    X_test = np.load('test_samples.npz')
    y_test = np.load('test_labels.npz')

    X_train, y_train, X_test, y_test = \
        X_train['arr_0'], y_train['arr_0'], X_test['arr_0'], y_test['arr_0']

    batch_size = 32
    nb_classes = 2
    epochs = 7

    X_train, X_test, y_train, y_test = prep_train_test(X_train, y_train,
                                                       X_test, y_test,
                                                       nb_classes=nb_classes)

    # 513x125x1 for spectrogram with crop size of 125 pixels
    img_rows, img_cols, img_depth = X_train.shape[1], X_train.shape[2], 1

    X_train, X_test, input_shape = keras_img_prep(X_train, X_test, img_depth,
                                                  img_rows, img_cols)
    print('Fitting model...')
    model, history = cnn(X_train, y_train, X_test, y_test, batch_size,
                         nb_classes, epochs, input_shape)
```
I successfully quantized the model using 

```
!vai_q_tensorflow quantize \
    --input_frozen_graph frozen_graph_depression_test.pb \
    --input_fn input_func.calib_input \
    --output_dir quantized_depression \
    --input_nodes input_1 \
    --output_nodes dense_2/Softmax \
    --input_shapes ?,513,125,1\
    --calib_iter 32
```

However, when I tried to compile the model I got this error:

VITIS_AI Compilation - Xilinx Inc.
[INFO] Namespace(inputs_shape=None, layout='NHWC', model_files=['quantized_depression_V2/quantize_eval_model.pb'], model_type='tensorflow', out_filename='./depression_V2_org.xmodel', proto=None)
[INFO] tensorflow model: quantized_depression_V2/quantize_eval_model.pb
[INFO] parse raw model :100%|█| 26/26 [00:00<00:00, 27134.09it/s]
[INFO] infer shape (NHWC) :100%|█| 33/33 [00:00<00:00, 5077.85it/s]
[INFO] infer shape (NHWC) :100%|█| 30/30 [00:00<00:00, 4038.81it/s]
[INFO] generate xmodel :100%|█| 30/30 [00:01<00:00, 17.32it/s]
[INFO] generate xmodel: /workspace/depression_V2_org.xmodel
[UNILOG][INFO] The compiler log will be dumped at "/tmp/vitis-ai-user/log/xcompiler-20221031-153814-245"
[UNILOG][INFO] Target architecture: DPUCZDX8G_ISA0_B4096_MAX_BG2
[UNILOG][INFO] Compile mode: dpu
[UNILOG][INFO] Debug mode: function
[UNILOG][INFO] Target architecture: DPUCZDX8G_ISA0_B4096_MAX_BG2
[UNILOG][INFO] Graph name: quantize_eval_model, with op num: 44
[UNILOG][INFO] Begin to compile...
[UNILOG][WARNING] xir::Op{name = max_pooling2d_1/MaxPool, type = pool-fix} has been allocated to CPU.
[UNILOG][Check Failed: bank_addr_end >= min_runtime_size][/home/xbuild/conda-bld/xcompiler_1606385139459/work/src/pass/passes/GenInst/AssignBankGroup.imp:1858][XCOM_ALLOCATE_BANK_FAIL][There is not enough bank space for the tensor]
*** Check failure stack trace: ***
Aborted (core dumped)

I don't get where the problem is?
Regards

skalade, 2022-11-28:
Hi there,

Sorry for the late reply. I'd recommend trying a the latest docker or the docker version compatible with the dpu-pynq version you are using.

If you get the same error with the latest docker, consider trying to reduce the size of the model... Maybe you're running out of RAM?

Thanks
Shawn

Afef00, 2022-11-28:
@skalade 
Thanks for the reply. OK I will try this

skalade, 2023-05-09:
Assuming the issue has been resolved since there has been no follow up.

closed 2023-05-09 by skalade