Xilinx/DPU-PYNQ #100: environement mismatch problem

jimmy-adams, 2023-05-08:
Hello Everyone!,
When is tried to complie the board related stream with **make board** command under the **boards directory**, the make will output an error shown the environment mismatch, the tutorial gives the flow for vitis 2022.1, but mine is vitis 2022.2.
I am not sure if the version difference will affect the final output stream?

Thanks alot!

skalade, 2023-05-09:
Hi @jimmy-adams,

The build scripts have only been tested with Vitis 2022.1 in this version of dpu-pynq, which is in line with Vitis AI 2.5. You could try to edit the check_env.sh script and change it to 2022.2, but I can't guarantee that the scripts will run successfully with your setup.

Thanks
Shawn

skalade, 2023-11-29:
Closing due to lack of activity.

closed 2023-11-29 by skalade