Xilinx/DPU-PYNQ #102: Multiple subgraph support

Rellek72, 2023-05-20:
Does this project have any support or planned support for models made of multiple subgraphs? I attempting to implement a Transformer architecture that consistently generates into 8-12 subgraphs, and line 177 of [dpu.py](https://github.com/Xilinx/DPU-PYNQ/blob/master/pynq_dpu/dpu.py) seems to put a hard limit of only graphs made of a single subgraph are allowed.

Rellek72, 2023-05-23:
Thank you for your help. Could you refer me to example code of using VART in dpu-pynq? When I try to use my code or examples from Vitis AI 2.5 I'm missing libraries such as `dpu` and `vitis_ai_library` needed to run VART in python.

skalade, 2023-05-24:
We don't really have examples like that, the reference would be ripping out what's in [dpu.py](https://github.com/Xilinx/DPU-PYNQ/blob/master/pynq_dpu/dpu.py). 

This snippet, for dpu_resnet50.xmodel should work if you run it a jupyter lab session in the same folder as the pynq-dpu example notebooks folder with the delivered .xmodel files. 
```
from pynq_dpu import DpuOverlay
overlay = DpuOverlay("dpu.bit")

import vart
import xir

# Get list of child subgraphs
graph = xir.Graph.deserialize('dpu_resnet50.xmodel')
root_subgraph = graph.get_root_subgraph()
child_subgraphs = root_subgraph.toposort_child_subgraph()

# Check what devices each subgraph is meant to be run on
for cs in child_subgraphs:
    print(cs.get_attr("device"))

# After this step you should be able to just re-use all the code in the resnet50 example notebook
dpu = vart.Runner.create_runner(child_subgraphs[1], "run")
```

Hope this helps get you started.

skalade, 2023-11-29:
Closing due to lack of activity.

closed 2023-11-29 by skalade