Xilinx/DPU-PYNQ #119: MNIST archives not available from LeCun's website

pcotret, 2025-03-25:
Trying this notebook: https://github.com/Xilinx/DPU-PYNQ/blob/master/pynq_dpu/notebooks/dpu_mnist_classifier.ipynb

The MNIST pip package will download data from Yann LeCun's website http://yann.lecun.com/exdb/mnist/

The website isn't available anymore, a PR in the MNIST package gives a mirror: https://github.com/datapythonista/mnist/issues/20

Maybe adding this somewhere in the Readme file ?

pcotret, 2025-03-25:
Ot it works if you manually download it to the temporary directory (which is `/tmp` by default). It will check this repository before trying to download those files.

smtalds-knt, 2025-05-22:
```python
import torch
from torchvision import datasets, transforms

# Define a transform to normalize the data
transform = transforms.Compose([transforms.ToTensor()])

# Download the MNIST training and test datasets
train_dataset = datasets.MNIST(root='./data', train=True, download=True, transform=transform)
test_dataset = datasets.MNIST(root='./data', train=False, download=True, transform=transform)

```
You can download like this to host machine. Then you can upload as zip these files to board.